jsanpe/c2c-64 #3: BOM Questions

ezcGman, 2024-01-22:
Hey there,

I'm currently collecting parts to build some of these awesome board (thanks for making these available!!).

I do have a question about this one:
> Ferrite_Bead	FB1,FB2,FB3,FB4,FB5,FB6	Inductor_SMD:L_0805_2012Metric

Do you have a bit more specs or an exact part number for these? LCSC (obviously) spits out a million results for just an inductor ;)

Thanks and greetings,

Andy!

jsanpe, 2024-01-22:
Hi,
thanks for reaching out. Sorry about the poor state of the project, I unfortunately have little time these days to maintain/improve it. 
Regarding your questions:
- For ferrites I used these: https://www.lcsc.com/product-detail/Ferrite-Beads_Sunlord-GZ2012D601TF_C1017.html but they seem to be out of stock nowadays. You should be able to find something with similar characteristics in lcsc (or elsewhere)
- For crystals, note that Y1 and Y6 are mutually exclusive. You should use one or the other, depending on availability. The freq is 28.636363Mhz. I have used this one in the past: C136080.
- The same applies to crystals Y2/Y3, you can choose either one. However, this crystal is not needed as the MPU internal oscilator is used by default. So, you can leave this (and the caps C46 and C47) unpopulated.
- RN1 is a resistor array. 33 ohm is recommended, but you can use something close to it if you don't find it. I actually have been using this part: C78520 which is 39ohm

I will try to use the plugin you recommend and push the output to github. I can't really commit to this, lot's of other stuff in my life right now.

jsanpe, 2024-01-22:
It was easier than I expected: https://github.com/jsanpe/c2c-64/tree/master/hardware/1R/1R3/bom

ezcGman, 2024-01-22:
> Hi, thanks for reaching out. Sorry about the poor state of the project, I unfortunately have little time these days to maintain/improve it. Regarding your questions:
>
> * For ferrites I used these: [...]

They're still in stock for me using that link, added them to the cart, thx!

> * For crystals, note that Y1 and Y6 are mutually exclusive. You should use one or the other, depending on availability. The freq is 28.636363Mhz. I have used this one in the past: C136080.

Oh ok. I just looked at the BOM, was there a note about that somewhere that I missed? I added the C136080 to the cart: https://www.lcsc.com/product-detail/Crystals_HELE-Harmony-Elec-X3S02863ADK1HA-X_C136080.html

> * The same applies to crystals Y2/Y3, you can choose either one. However, this crystal is not needed as the MPU internal oscilator is used by default. So, you can leave this (and the caps C46 and C47) unpopulated.

Ah ok. So that one is not even used at all? So there is really no need for it? I might just add it, because it's just a few cents, if there is the slightest chance for it to make any sense :)

> * RN1 is a resistor array. 33 ohm is recommended, but you can use something close to it if you don't find it. I actually have been using this part: C78520 which is 39ohm

Aaaaah yeah 🤦 That makes sense, thanks! LCSC doesn't have it in stock, will get it somewhere else.

Will continue adding everything to the cart, thanks already!!

Greetings,

Andy!

ezcGman, 2024-01-22:
Some questions on the fuse:
* Guessing 6v is fine, as everything is 5v on the board?
* Is the 1.5A the tripping or hold current? Guessing it's trip current?

Thanks!

jsanpe, 2024-01-23:
For fuse I used this: C311043. 6V is a bit tight, but you can always replace it if it blows too easily. At 1.5A you should use a fast fuse. The board consumption should be in the few mA range (300mA tops) but it's always higher when you switch it on, that's the reason I allow for some extra amps. You may want to go for a slower fuse with lower amp rating.

About Y2/Y3 the current firmware does NOT use them. The internal atmega oscillator is accurate enough to allow for UART to work fine at slow speeds. For all other intents and purposes the accuracy of the oscillator is simply irrelevant, as the MPU is only there to configure the Video encoder and decoder ICs. My recommendation: do not populate.

ezcGman, 2024-01-23:
One more sadly :D Do you by any chance have a link to the buttons you used? There are a quadrizillion buttons out there and finding the ones you used with the spacing... it's close to impossible :D Thanks!

jsanpe, 2024-01-23:
> One more sadly :D Do you by any chance have a link to the buttons you used? There are a quadrizillion buttons out there and finding the ones you used with the spacing... it's close to impossible :D Thanks!

These little f$ckers are hard to find, tell me about it. Try these: C961730

ezcGman, 2024-01-24:
Thanks for your extensive help!

I've ordered everything, most from LCSC and some connectors from AliExpress. I'm attaching the CSV of my LCSC order for people to easier re-order the components. Also, there are the items on AliExpress I ordered:

* https://www.aliexpress.com/item/1005004682978457.html
* https://www.aliexpress.com/item/4001206395694.html
* https://www.aliexpress.com/item/1005006152724809.html
* https://www.aliexpress.com/item/1005003406780797.html

[LCSC_Exported__20240124_210810.csv](https://github.com/jsanpe/c2c-64/files/14038775/LCSC_Exported__20240124_210810.csv)
**ATTENTION:** A few caps are missing from this CSV, as I still had them in stock here. Forgot which ones these were. I'll update this post as soon as I receive the order, so I can easily see what's missing, so what I need to get out of my personal stock.

ezcGman, 2024-01-26:
Will do! I've updated the CSV to now reflect the full components list, so people can just use that :) Closing this for now :)

closed 2024-01-26 by ezcGman

whytheheckme, 2024-05-07:
@ezcGman Did you ever get this up and running? Do you have spare parts you'd like to sell (particularly a PCB?)

ezcGman, 2024-05-08:
> @ezcGman Did you ever get this up and running? Do you have spare parts you'd like to sell (particularly a PCB?)

Jepp, I did: https://github.com/jsanpe/c2c-64/issues/5 

And yeah, I do have complete (just missing the screen, but that is easy to get via AliExpress) left over kits I can sell. Because I bought parts for 5 kits directly, the price per kit is actually super low, it's just below 20€! I can spare one, if you want it? I'm in Germany, so depending on where you are, shipping costs may be the same as the price for the kit ;)

whytheheckme, 2024-05-08:
> And yeah, I do have complete (just missing the screen, but that is easy to get via AliExpress) left over kits I can sell. Because I bought parts for 5 kits directly, the price per kit is actually [...]

Amazing! I just finished pricing out all of the parts and the PCB, and I really don't need more than one, so I think it would be worth it. I live in the US but have an address in Denmark you can send it to where I will be in about 10 days (I found I could generate a label for 12€ but you may be able to get it even cheaper!)

Can you drop me a line? jake.at.spam (at) gmail
Thank you!!

whytheheckme, 2024-05-10:
@ezcGman Just wanted to follow up, I'd love to connect and buy a kit set from you.