jsanpe/c2c-64 #4: Question about orientation of ICs

ezcGman, 2024-02-14:
Hey there (again) :)

I was just about to solder one of these and then realized that the PCB is missing the usual dot indicators to tell about the orientation of the ICs. I saw that the silkscreen/graveyard on one side is open instead of closed. Is that the orientation indicator?

Talking about this:
[image]

Thanks and greetings,

Andy!

jsanpe, 2024-02-15:
Yup, Kicad's default footprints indicate pin number 1 using the open angle, as you mentioned

ezcGman, 2024-02-15:
Ou, never realized. I also use KiCAD but that I never saw.
Thanks!! Off to the workbench then \o/

closed 2024-02-15 by ezcGman