jsanpe/c2c-64 #5: Failing to program via UART (and maybe also to transfer the bootloader)

ezcGman, 2024-02-16:
Hey there,

it's me again... I finished building the board and everything looks fine, but I struggle a bit to get it flashed via UART. And now I'm not even sure if I did everything right for the fuses and bootlader.

I'm using this for reference: https://github.com/jsanpe/c2c-64/wiki/Programming-the-C2C-64

First, I don't have an Arduino DUE, but I googled a bit and it seems that you can also use an ESP8266 or ESP32 (which I work with A LOT) to achieve the same. So I successfully flashed the ArduinoISP sketch (it just compiles naturally, I just needed to change some pins) to a random ESP8266 (Actually a NodeMCU) I had using ArduinoIDE. Then (because you're using/referencing a platformio.ini file) I opened my VS.Code which I also use a lot to flash ESPs using platformio, so I'm pretty familiar with that.

As I'm on Windows, I opened the "fuses_1R3" task, changed the upload_port to "COM4" and I could successfully run the "Set Fuses" and "Burn Bootloader" platform tasks (which is I guess what you're referencing in your guide). I then got out my FTDI dongle utilizing a FT232BL chip (https://www.aliexpress.com/item/32967622307.html) and tried to flash the software via UART, but I always get the same error: `avrdude warning: attempt x of 10: not in sync: resp=0x00`

Yes, sadly my dongle doesn't have an easy way to wire RST, but it has the pin exposed, so I'm touching it manually. But no matter how often I try, I always get the same error. I also tried simply bridging RST of the c2c to GND manually but it never works. No matter how often I try and when... It never works. I even tried reversinh TX/RX, just in case: Still nothing.

Then I thought, after reading your guide 100th time, did I actually flash the bootloader? I just used the platform task "Burn bootlader". I then got my ESP wired again, validated I can still execute "Set Fuses" and then just hit "Upload" (still under the "fuses_1R3" task), but it first failed with not finding the "SSD1306Ascii" library, because it's missing in the "fuses_1R3" task. Added it and hit "Upload" again. It compiled but then getting the same error as I got via UART: `avrdude warning: attempt x of 10: not in sync: resp=0x00`

I'm a bit lost what I'm doing wrong... Do you maybe have an idea what's happening? I don't think my board is broken, right? Otherwise I wouldn't be able to execute the platform tasks "Set Fuses" and "Burn Bootloader" successfully, right? :/

Thanks so much again and greetings,

Andy!

jsanpe, 2024-02-17:
Hi,

congrats for having make it this far! Not an easy task, I understand you may be impatient now but it's better to just take your time in the programming phase as well.

Before anything, just in case you haven't, take some measurement of voltages at the output of the 3 regulators to make sure everything is in check. The big one should output 3.3v, the two mini ones on the back side 1.8v. Reduce your PSU amp to 0.5A, 2A will blow your fuse (or more) if there's a short now or in the future.

Burning the bootloader requires just executing the "burn bootloader" task in the fuses_1R3 environment. No need to "upload" or "set fuses". Looking at the output of the command it seems like your ESP flasher did the work just fine. Not sure what running the "upload" task of the fuses environment does to be honest. Perhaps it's uploading the actual firmware instead of the bootloader. This may not work well in practice, as the firmware is not loaded in the right address to be executed at startup most likely. I would repeat the burn bootloader process following this instructions: go to environment fuses_1R3 and run only the "burn bootloader" task.

The output of avrdude suggests that the bootloader is not entering in flashing mode. This could be related to the RTS signal not being timed well, the bootloader is very picky in this regard. The RTS needs to be followed by the stream of data to flash, if not timed correctly the whole process won't work. Hence, it's recommended to have a supported programmer, otherwise flashing is a matter of luck more than anything. I'm using [this one](https://www.amazon.es/DSD-TECH-Adaptador-FT232RL-Compatible/dp/B07BBPX8B8/ref=sr_1_6).

Other things to check:
- These qfn packages are devilish, sometimes pads are not properly soldered. Try measuring resistance from tx,rx and rst to gnd. It should be in the MOhm range.
- The programming circuitry mainly involves C3 and R2. Check these two are correctly soldered.
- I notice you have populated the SW1 switch. This one is a reset. You can use this to manually send the reset pulse instead of the RST pin. Remember, it needs to be timed really well with respect to avrdude. Experiment with it, try pressing sw1 just before or just after running avrdude.

Keep me posted with any further results!

Good luck,

Jose

ezcGman, 2024-02-17:
Hey Jose,

thanks for your quick answer, appreciated!! And thanks for the compliment! I did some tricky boards in the past, but that one was certainly one of the most challenging, especially with 0402 components. The rest was tricky, but did that before :) Impossible basically without a microscope, imho! Or unnecessarily complicated!

Yeah, I of course did some extensive poking with the multimeter before connecting / checking for bridges, inspected under microscope and so on an are 99% certain there are none. First thing after powering on was checking the LDOs (both the AMS1117 and the TLV700 on the back) and they spit out what they should. So all good here.

I'm not concerned about blowing the fuse, as it's a PTC / resettable fuse. It will blow, reset and be fine again :) Hold current is 750mA and tripping current is at 1.5A. But still good advice, I lowered the PSU to 1A for now.

* Measured RX & TX against GND: 4.2 / 4.3MOhm. So looks fine
* R2 measured 4.6kOhms (really touching the solder blobs, not the resistor, so making sure it's properly soldered. So also fine...
* Goooood point! I totally forgot about the buttons. That sadly also didn't help, but the error at least a little bit changed from `0x00` to: `avrdude stk500_getsync() warning: attempt 10 of 10: not in sync: resp=0xa0`...

I now switched to the console, running avrdude directly / outside VS.Code. That's my command and output:
```
C:\Users\andy\.platformio\packages\tool-avrdude>avrdude -v -p atmega328pb -C C:\Users\andy\.platformio\packages\tool-avrdude\avrdude.conf -c arduino -b 38400 -D -P COM5 -U flash:w:C:\Users\andy\Documents\Sources\c2c-64\.pio\build\Board_1R3\firmware.hex:i

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\andy\.platformio\packages\tool-avrdude\avrdude.conf

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 38400
avrdude stk500_getsync() warning: attempt 1 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 2 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 3 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 4 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 5 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 6 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 7 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 8 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 9 of 10: not in sync: resp=0xa0
avrdude stk500_getsync() warning: attempt 10 of 10: not in sync: resp=0xa0
avrdude main() error: unable to open programmer arduino on port COM5

avrdude done.  Thank you.

C:\Users\andy\.platformio\packages\tool-avrdude>
```

I'm wondering: Is the baudrate and also the upload protocol of "arduino" correct? If I see that right, it's basically a re-config of "stk500"?

If you can spot an issue with that, I might just get another FTDI tool. I read a bit and generally it seems the FT232RL (that is used on your flasher) generally gets "better feedback" than the FT232BL I have on mine.

Thanks again for double checking and greetings,

Andy!

jsanpe, 2024-02-17:
I reflashed my C2C to capture the exact avrdude command and the logs. You'll find the output below.

To answer your questions, both baudrate and protocol are fine. Keep testing it out, I would recommend resetting the board using sw1 at the same time you press Enter on the command line using the avrdude arguments in your previous post. Try several times, vary your timing slightly, you should get a successful programming at some point. If you don't there may be something else going on. If you have a scope try captuing the reset line to the MCU using C3 pads. 

If you are unable to flash using the UART I think it should also be possible to flash the firmware using the ICSP header, but I would need to investigate it a bit more. Never used that option before.

```
avrdude -v -p atmega328pb -C /home/jsan/.platformio/packages/tool-avrdude/avrdude.conf -c arduino -b 38400 -D -P /dev/ttyUSB0 -U flash:w:.pio/build/Board_1R3/firmware.hex:i

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/jsan/.platformio/packages/tool-avrdude/avrdude.conf
         User configuration file is /home/jsan/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 38400
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino for bootloader using STK500 v1 protocol
         Hardware Version: 3
         Firmware Version: 8.0
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)

avrdude: processing -U flash:w:.pio/build/Board_1R3/firmware.hex:i
avrdude: reading input file .pio/build/Board_1R3/firmware.hex for flash
         with 18630 bytes in 1 section within [0, 0x48c5]
         using 146 pages and 58 pad bytes
avrdude: writing 18630 bytes flash ...
Writing | ################################################## | 100% 9.40s
avrdude: 18630 bytes of flash written
avrdude: verifying flash memory against .pio/build/Board_1R3/firmware.hex
Reading | ################################################## | 100% 9.31s
avrdude: 18630 bytes of flash verified

avrdude done.  Thank you.

============================================== [SUCCESS] Took 21.41 seconds ==============================================```

ezcGman, 2024-02-17:
I sadly don't have a scope. I just order another flasher that properly exposes the RTS pin and see how it goes :(

Thanks for all the help so far!

jsanpe, 2024-02-17:
Let's see if you have better luck with the new FTDI dongle. It's a good investment, lot's of potential future uses. I'm still positive is relatively doable to get this done by hand, i.e. manually resetting the mcu with the right timing. In any case, I'll check how to flash directly though the ICSP header, it's probably not too difficult to achieve but not as practical as the UART solution (especially during development).

ezcGman, 2024-02-18:
Yeah I tried it MANY times by now and zero luck. I tried a lot of timings, also inbetween the retries... Nothing :/

jsanpe, 2024-02-18:
A couple of quick things to try:
- reduce all instances of 38400 in platform .ini with 9600. You will need to rewrite the bootloader using the Burn Bootloader command. This will reduce the UART speed used by the flashing procedure, which depending on the status of the internal oscillator of your 328PB may help.
- In your platform.ini file replace `board_hardware.uart = uart0` with `board_hardware.uart = no_bootloader`. Then use "Upload" in the Fuses_1R3 environment. I think this should upload the firmware directly using your ESP as interface and place it in the right flash location for the MCU to boot it directly. I haven't tested this myself, but it's my conclusion after looking at the documentation here: https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md

ezcGman, 2024-02-19:
First, thanks on digging out the instructions on how to program using the ICSP header and tips on how to get it working via UART from your previous post! I gonna test that later today with my current FTDI flasher, but I also have the exact one coming that you're using. Let's see how it goes.

But another thing I just realized while sitting in a boring call and staring on the PCB:
> * The programming circuitry mainly involves C3 and R2. Check these two are correctly soldered.

I discovered that C3 and R2 are not connected to RX/TX/The UART circuit, but instead to the SCK pin. RX and TX go straight to the MCU. Just wanted to double check if that was an oversight in the PCB design, or you just misremembered (no blame intended!! I just saw that by accident :D)

Thx again!

ezcGman, 2024-02-19:
The flasher you use just arrived and it sadly also doesn't work :( Same error as above (0x00). I tried reversing TX/RX, I tried using RTS and CTS: Both don't work :(

I'm clueless what it can be... I also multimetered the pin header pins and the pads on the atmega: they have continuity, so... I have no clue :D
I'll try lowering the baud rate and see if that will help...

jsanpe, 2024-02-19:
Hi! Sorry to hear the new flasher is not working :( I was starting to think that the programmer wasn't the issue after the many tests you've been making. 

Answering your question about C3 and R2, this is the reset circuit that is triggered by the RTS pin. Once the reset happens, the bootloader enters into action and gives you 1 or 2 seconds to start sending the data through uart. If this reset circuit does not work, then the bootloader never listens for a new firmware. TX and RX are connected directly to the MCU. That's ok.

I've been checking out the MiniCore project and it seems that there has been a change in the bootloader used. I wonder if this could be the issue. This can be tested quite quickly, change the line:
`upload_protocol = arduino` in env Board_1R3 to `upload_protocol = urclock`
Or you can force the use of the previous bootloader (optiboot) by setting
`board_hardware.uart = optiboot` on the Board_1R3 env

Another quick check to verify if the bootloader is running is to place a led with a current limiter resistor on the RST pin (right above the TX pin). The bootloader should either blink it or completely switch it on on boot.

Hope this helps, and sorry for all the trouble. If this does not work, then I will recommend to flash through the ICSP header.

jsanpe, 2024-02-19:
Hi again,

I've been doing some more reading and after inspecting the firts log you sent me I've found this message:
```
Building in release mode
Using bootloader image:
C:\Users\andy\.platformio\packages\framework-arduino-avr-minicore\bootloaders\urboot\atmega328pb\watchdog_1_s\autobaud\uart0_rxd0_txd1\led+b5\urboot_atmega328pb_pr_ee_ce.hex
```
which means you have flash urboot bootloader. Again, this is new for MiniCore (which is the backbone of C2C) and was introduced in version 3.0.0 in Oct 2023. So I was not aware of this change. I think this is what's behind the issues you are facing. Replace the `upload_protocol` as I told you in the past message, that should fix the issue straight away. 

There's more info about the new bootloader [here](https://github.com/stefanrueger/urboot) and these is the avrdude command they recommend using:
`avrdude -c urclock -p m328p -b 115200 -P /dev/ttyUSB0 -U flash:w:mysketch.hex:i` 

Again, I think this should solve the issues you are facing. I really hope this time it helps.

ezcGman, 2024-02-19:
HOLY MOTHER OF GOD, IT WORKED!! :D YOU ARE A GENIUS!!! \o/ Even with my old flasher \o/
THANKS SO MUCH!!!!

So the only changes I now ended up going to the current version of `platformio.ini` in your repo is:
* Under `[env:Board_1R3]` setting this `upload_protocol = urclock`
* And for me / under Windows under `[env:fuses_1R3]` setting `-PCOM4`

And confirming my packages and versions for people to come:
```
PACKAGES: 
 - framework-arduino-avr-minicore @ 3.0.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
```

Can't wait to test the board now!!

jsanpe, 2024-02-20:
Nice! Thanks for having the patience mate, and sorry for all the trouble. This change of bootloader in MiniCore was something I wasn't aware of. And it comes with some breaking changes :( It seems to be a much better/modern bootloader, so the change is for good. But you were the first one to face it. 

I will update the platform.ini file in the project with the new upload protocol, which apparently is backwards compatible with arduino.

Enjoy your new C2C!

ezcGman, 2024-02-20:
Absolutely no issue, this is a DIY board, these things can and may happen! Thanks to you for the extensive support here!!

I'll try to give it a test this week, packed week this is! I'll follow up with a tiny post/instructions how I used an ESP to flash the bootloader, instead of an Arduino Due. Think this is prettx useful, as everybody probably has an ESP8266/ESP32 somewhere. And if not: It's not even 5€ :)

Later,

Andy!

jsanpe, 2024-02-20:
Thanks Andy, if you can write those instructions I can add them to the wiki along with the DUE so people have more options. Let me know if that's ok!

Jose

jsanpe, 2024-02-20:
Neat, thanks so much! I'll add this to the wiki and ensure proper credit is given.

I have already updated the platform.ini file in the repo. It should work fine for both optiboot and urboot now.

closed 2024-02-20 by ezcGman

jsanpe, 2024-02-20:
Wiki updated! https://github.com/jsanpe/c2c-64/wiki/Programming-the-C2C-64

ezcGman, 2024-02-20:
Thanks!! Also for the credits!!

Just saw there is a double "the" in the text: "... that runs the the Arduino ..." :)

ezcGman, 2024-02-20:
It works!! \o/ Thanks for this fantastic project!!
[image]
[image]

jsanpe, 2024-02-20:
Congrats! N64 looks fantastic through composite, and that leaves you s-video for older systems. For instance, SNES looks awesome through s-video. One C2C, two consoles!

Remember to get a case asap, too many parts to short on that board :D

ezcGman, 2024-02-20:
I actually have another N64 RGB modded, so I play N64 using RGB-SCART through the OSSC. Same for SNES, as my PAL SNES still has RGB out natively ;)

In the end, all consoles (and I have quite a few...) work through either RGB-SCART, Component or HDMI already. So I currently don't REALLY NEED the c2c, but I wanted it, as I have no way of connecting composite or S-Video sources to my display. Now I have, if I ever need. The only really need was that N64 which I didn't mod but wasn't sure if it works. And here we go: It does ;)

khari05, 2024-05-26:
I was having a similar issue with programming over UART. I never actually solved it, but I got it working using ICSP. Wiring the ICSP header on my generic ft232r programmer from MicroCenter, and setting the upload protocol to the custom avrdude config I had to add to be able to use it properly, got it flashed properly:

[image]