jsanpe/c2c-64 #7: Looking for a pick and place file

khari05, 2024-04-22:
I'm trying to make my own board and thought I'd try to make JLC assemble the SMD components on the top side. In KiCad, there's a fabrication output option called "Component Placement" and you can set the output format to CSV. If you share the KiCad project, I can tinker with it to get only the components I need. If not, I'd appreciate a component layout CSV for the front.

jsanpe, 2024-04-23:
Hi, sorry for the late response. Really busy times for me. I have found the pos file I think I used back in the day. Give it a try, it should work fine.
[composite2rgb-top-pos.csv](https://github.com/jsanpe/c2c-64/files/15082444/composite2rgb-top-pos.csv)

cheers

khari05, 2024-04-24:
Thank you so much for your reply! This will save me so much time.

I have a few questions about the BOM:
- does R1 need to be 169Ω? I'm trying to mostly use basic components from JLC, and it doesn't seem like I can find a similar value. Might end up purchasing them separately and soldering it by hand
- Is it okay to replace R4,R5,R6 with 0603 resistors? They look like termination resistors, so I don't think any current will run through them.

Other than that, I think I am just going to hand solder anything I can't find as a JLC basic component other than the main ICs since the $3 reeling fee per extended component can be hefty

jsanpe, 2024-04-27:
Hi,

about R1, it is part of the video encoder PLL circuit along with C1 and C6. All these are pretty critical parts, and it is recommended to use high quality components with little tolerances for them. 

R4-R6 can be replaced with 0603 components without any issue.

cheers!

khari05, 2024-05-16:
Thank you for helping! I got the partially assembled boards in the mail and my first one will be ready for testing once my hotplate arrives and I can rework RN1 (I accidentally bridged the contacts on it)-- I really should've paid to not have to hand solder it.

My total for boards + components was ≈$40 USD per board because of shipping and since I paid the $3 setup fee per component for the major ICs to be pick-and-placed by JLC

closed 2024-05-16 by khari05